amnemonic/Quansheng_UV-K5_Firmware #7: uvmod_kitchen update ?

goopaga, 2023-06-09:
Hi Adam @amnemonic, let me start with a huge thank you for the effort you are putting into this great development of a surprisingly capable radio (for the price). I understand that this is team work with other brillant people, kudos to everybody involved !

I'm trying to coock my first firmware mod with the files from the "uvmod_kitchen" repository. However looks like the file "mod_enable_tx_50to850.py" is a previous version with "limited" band.

Would it be possible to update the repo with the latest file (18 to 1300 mhz) ?

Thanks
73' IU4MJW

amnemonic, 2023-06-09:
Hello @goopaga and thank you for your kind words!
Mod named `mod_enable_tx_50to850.py` should/can be used together with `mod_custom_freq_ranges.py` to also extend frequency ranges for each band. You have to remember that default input file is stock firmware. 

The purpose of  `mod_custom_freq_ranges.py` mod is to give ability to user to select which ranges should be covered by each frequency range of quansheng radio or to extend them at all. By default, file contains frequency ranges exactly like in stock firmware. If you open file in text editor you will see these lines:

```python
# change below sets to new ones, values are in Hz
new_freq_low_limit =   [50_000_000, 108_000_000, 136_000_000, 174_000_000, 350_000_000, 400_000_000, 470_000_000]
new_freq_high_limit =  [76_000_000, 135_999_900, 173_999_900, 349_999_900, 399_999_900, 469_999_900, 600_000_000]
```

Here you can change low and high limit for each frequency band. The underscore `_` symbol is omitted by python interpreter and is added only for better readability. So for example, if you want to fill the gap between 76 and 108Mhz then in second array change first limit from `76_000_000` to `107_999_990` or if you want to extend above 600MHz then change last limit from `600_000_000` to `1300_000_000`. Please keep in mind that different ranges are demodulated slightly different inside BK4819 chip, and some range have enable AM demodulation while other not. That's why there is no universal way to _just enable full frequency range_ from 18 to 1,3GHz.

closed 2023-06-11 by amnemonic

marusc1, 2023-06-29:
Can frequencies down of 28 and up of 1300 be set for testing real hardware limits?