amnemonic/Quansheng_UV-K5_Firmware #16: Customize TX DISABLE

arturmaciag, 2023-06-13:
Is it possible to customize TX disable on any frequency ranges? 
Currently, mod_enable_tx_50to850 unlocks all bands.

amnemonic, 2023-06-13:
Hello @arturmaciag,
just pushed new commit with mod: [mod_disable_tx_completely.py](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/uvmod_kitchen/mod_disable_tx_completely.py) 
It is tested and works as you described. 
I have to work on mod which disables all ranges except for example PMR446 though.

closed 2023-06-13 by amnemonic

arturmaciag, 2023-06-13:
Thank you for your commitment.
I am interested in blocking the TX of only one band (AirBand).

BTW:
I don't see any bit pattern masking in enable or disable TX (register address 0x180E). How do you know which bits to set?

amnemonic, 2023-06-14:
Thank for kind words. That is not bit pattern, it is ARM cpu instruction to return from function with `False` in register `r0` informing in this way that this frequency is not allowed for TX and `0x180E` happens to be the best place to decide where to jump in the code to get desired return value.

arturmaciag, 2023-06-19:
Thank you for your AirBand Tx Disable feature! Good job! 
I have the last question in TX disable issue: how to unlock TX lower than 50MHz? @Tunas1337 has a mod to unlock 18 to 1300MHz. Your mod kitchen allows unlocking 50 to 850 MHz.

Tunas1337, 2023-06-19:
It's the same thing, really. Also, @fagci was the one who made that mod.
I'll see if I can make it so that you can customize the TX range, but it would take time.

RE3CON, 2023-08-07:
mod_enable_tx_50to850.py
enable tx from 50 to 600MHz instead of 850MHz ???
by changing bit at address to 5de0