amnemonic/Quansheng_UV-K5_Firmware #17: Use hidden menu (switch on with PTT+Sidekey1 pushed) to enable/disable patches

RE3CON, 2023-06-16:
switching the radio on with PTT and Sidekey 1 pushed appears in custom firmware redunant entries. There functions are bypassed with custom firmware. That's 350TX on/off, 200TX on/off, 500TX and 350Enh. TX on/off, F-Lock too? 
Giving this another purpose. Changing it to Display Invert on/off, TX Airband on/off, TX on/off, Mic Gain +16 on/off etc... allows the user more controll.
Enter to VOL in menu (voltage) makes no changes, should/could adjust the real battery Voltage?

fagci, 2023-06-16:
Confirm if I got you right. F-LOCK has now "abcd"

RE3CON, 2023-06-16:
No I mean use these 4 or 5 functions and change them usage maybe 200TX on/off to TX Air on/off, LCD invert on/off, TX all Freq. On/off. 
F-Lock I don't know for what it is but VOL shows the Voltage no changes when go into the settings. Could be adjusting the tollerance it shows by Voltage. By some devices it's up to 0,8 Volt +/- wrong.
Hook into these Functions to enable disable some patches. May be it is possible?

marusc1, 2023-06-29:
The idea is great. Why do you mention the vol readout at same issue? It makes it difficult to understand.

RE3CON, 2023-07-03:
These new entries under F-Lock like 123, abcd9, 9, 12345 only by patched firmware what do they do?

RE3CON, 2023-07-03:
> The idea is great. Why do you mention the vol readout at same issue? It makes it difficult to understand.

Yah VOL in menu stand for voltage not volume.

closed 2023-11-18 by amnemonic