amnemonic/Quansheng_UV-K5_Firmware #19: How the reverse engineering is done？

cheng3100, 2023-06-21:
After reading some script in the repo. I understand how the firmware is pack/unpack and how it is modified. But I'm really curious about， how it is found？
For example, how do you know the raw binary is xor by a special sequence? And how do you know the sequence?
In the script https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/uvmod_kitchen/mod_enable_tx_50to850_except_airband.py, how did you guys know that at the offset 0x1804 is the code that limit the tx freq value so that you can replace the asm code with a custom shell code? Is that really all done through  reading the disassemble code?

fagci, 2023-06-21:
One way to get xor key is to bruteforce it by known strings in binary.

cornz38, 2024-02-25:
By magic.. At least as far as i am concerned. Just pure magic and sorcery.