amnemonic/Quansheng_UV-K5_Firmware #29: Tx range limited using uvmod kitchen. 18 to 1300 not possible

marusc1, 2023-06-29:
I was using uvmod kitchen. My intention was to leave with with full tx range but it appeared impossible. The transmitter does not work below 50 and above 600 MHz. Is there a way to to set it to at least 18 to 1300? 
I tried to start with modded firmware ( 18to 1300) without enabling any tx modes and it works for some mods like mic gain and noaa freq replacing. But than the frequency ranges setting mode does not work as it  checks original input ranges and reports error.  What is this check done for?

Szetya, 2023-06-30:
I have similar problems. :)

Szetya, 2023-07-02:
Solved! TX range lower limit is stored in 0x150C (32bit), upper limit in 0x1510 (32bit).  I wrote a very simple and insecure script that I don't want to share here. @amnemonic can you help me with a working script?

amnemonic, 2023-07-02:
Thank you @Szetya for finding these values. I've added new script in commit 16c76a2.

closed 2023-07-02 by amnemonic