amnemonic/Quansheng_UV-K5_Firmware #31: mod_more_freq_steps is flawed

whosmatt, 2023-06-30:
https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/0804a92ee9951a31a48e0f9e977b4cc36d87c038/uvmod_kitchen/mod_more_freq_steps.py#L35C1-L38C28

Due to (presumably) compiler optimization, the constants for the array sizes are also used multiple other times in the firmware and cause unintentional changes. 
So far I noticed that other menu item array sizes also increased, such as backlight duration (ABR) as well as F-LOCK. Going out of bounds with one of the lower menu items that have text options instead of just a number can freeze the firmware. 

A potential solution might be adding new constants similar to how the new array was added and changing the references to those.

whosmatt, 2023-07-15:
i added an explanation and customizability in #43 

making backlight duration completely independent from freq steps requires reverse engineering of the menu functions to split off the ABR part into a seperate function.

amnemonic, 2023-07-15:
Awesome, just merged your PR. Thank you for your work 🥇

closed 2023-07-15 by amnemonic