amnemonic/Quansheng_UV-K5_Firmware #34: Unlimited time for the close field monitorin/frequency meter function.

marusc1, 2023-07-04:
There is a function F+4 that brings the radio in strong signal wide band search mod. If a transmission is received it displays the frequency and ctcss/dcs. A nice feature when observing somebody near transmitting with his radio. But it has to be quickly activated when seeing/supposing a transmission. It is because the function stops after few seconds.
A mode would be great to make the function stay active until it receives/catches something or is deactivated by the user.

amnemonic, 2023-07-07:
Please use this mod: [mod_disable_freq_copy_timeout.py](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/uvmod_kitchen/mod_disable_freq_copy_timeout.py)
It disables counter totally and function run indefinitely or until you press exit. Please check if result is what you are expecting.

marusc1, 2023-07-14:
It works. Thank you for your work. 
There is a problem though. Switched to search mode the radio stays in it indefinitely and that is fine. The display shows SCAN. After catching a frequency but without ctcss/dcs it is displayed and the display still shows SCAN.  This stays until a code is received and displayed. The displays shows SCAN CMP and it cancels the readout after few seconds. You need to hurry to read it.  And you have to show up the radio. Could you please disable this timer too?

marusc1, 2023-07-23:
Why did you mark it as completed?

amnemonic, 2023-07-23:
Ok, reopened to find this second variable which waits for CTCSS code :)

marusc1, 2023-07-23:
Waiting for ctcss is ok. Has no timeout. Problem is the one that starts after the tone is found and resets the readout.

marusc1, 2023-08-28:
It can approached a different way. By disabling active ctcss search in the F>4(FC) function. So the radio could after finding a frequency stay in fake ctcss search forever. The tone search if wanted  can be than activated by F>*(Scan).