amnemonic/Quansheng_UV-K5_Firmware #44: Tried to dump the latest firmware but it failed

CloverGit, 2023-07-16:
I recently purchased a UV-K5 and noticed that it is running firmware version _k5_2.01.27_ (not released on qsfj.com). I'm sure it's the latest firmware because I purchased the device from a retail store where just a few kilometers from Quansheng company. Using openocd and `uvk5.cfg` configuration file I can connect to my device normally, Connecting to the debugger only enters the HardFault state, even use `reset run`, but when I dump the firmware using the `dump_image flash.bin 0x0 0x10000` command I find that the saved file is filled with 0x00. I suspect it's because the chip has read protection turned on, but I can't be sure of that.
[image]

amnemonic, 2023-07-16:
Hello. It would be nice to have this firmware. We will try to figure something out. What hardware/programmer are you using for dumping?

CloverGit, 2023-07-16:
> Hello. It would be nice to have this firmware. We will try to figure something out. What hardware/programmer are you using for dumping?

I've tried dumping with ST-Link V3 SET and CMSIS DAP with the same result, programmer is openocd 0.12.0+dev-00211-g30b0e9af8 (2023-05-20-01:36)

amnemonic, 2023-07-16:
ok, so lets start from the beginning, and hope that it is not locked :D

This is how I was able to dump soft:
- Radio enabled in flashing mode (press PTT during power on) 
- battery inserted and pin `VCC` of programmer not connected, only `GND`, `SWDIO` and `SWCLK` connected according to below pinout
- lately we added `gdb_breakpoint_override hard` to `uvk5.cfg`, please remove this line and then try again
- to check if you can dump any data at all try to dump ram memory by invoking `dump_image ram.bin 0x20000000 0x4000`
- also you can play with `adapter speed` parameter however from my experience if speed is to high you will get error in openocd, not just zeroes in file

[image]

CloverGit, 2023-07-16:
> 好吧，让我们从头开始，希望它没有被锁定:D
> 
> 这就是我能够转储软件的方式：
> 
> * 闪烁模式下已启用无线电（开机时按 PTT）
> * 电池已插入，`VCC`编程器的引脚未连接，仅`GND`按`SWDIO`以下`SWCLK`引脚排列连接
> * 最近我们添加`gdb_breakpoint_override hard`了`uvk5.cfg`，请删除此行，然后重试
> * 要检查是否可以转储任何数据，请尝试通过调用转储 RAM 内存`dump_image ram.bin 0x20000000 0x4000`
> * 你也可以使用`adapter speed`参数，但是根据我的经验，如果速度太高，你会在 openocd 中得到错误，而不仅仅是文件中的零
> 
> [image]

Thank you, after trying I have successfully dumped my firmware. 

> ok, so lets start from the beginning, and hope that it is not locked :D
>
> This is how I was able to dump soft:
>
> * Radio enabled in flashing mode (press PTT during power on)
> * battery inserted and pin [...]

Thank you, after trying I have successfully dumped my firmware. I realized that the reason why the previous operation failed to dump was because the programming cable was still connected to the computer while I was doing the dump, and after unplugging it, everything worked fine!
[image]
[image]

CloverGit, 2023-07-16:
> ok, so lets start from the beginning, and hope that it is not locked :D
>
> This is how I was able to dump soft:
>
> * Radio enabled in flashing mode (press PTT during power on)
> * battery inserted and pin [...]

Thank you, after trying I have successfully dumped my firmware. I realized that the reason why the previous operation failed to dump was because the programming cable was still connected to the computer while I was doing the dump, and after unplugging it, everything worked fine!

I added that firmware to the latest pullrequest.

[image]
[image]

amnemonic, 2023-07-16:
Great! Thank you for your PR! #45

closed 2023-07-16 by amnemonic

troilus, 2023-07-16:
@CloverGit Is there any new features in this firmware?

CloverGit, 2023-07-17:
> @CloverGit Is there any new features in this firmware? I noticed Aleks says in 27, the quality of air band reception has changed. Heard weaker, but more legible.

So far I haven't noticed any new features with this firmware. Unfortunately, the signal from the airport was so bad that I could hardly feel the difference in reception quality.