amnemonic/Quansheng_UV-K5_Firmware #50: Can‘t get a output file from uvmod_kitchen

Noble114514, 2023-07-19:
Hello, I am BH4GCN, an HAM from China. And I am trying to customize a version of UV-K5 firmware by using uvmod_kitchen. 
I installed python and modified `build.bat` , and then ran it in the cmd window. The output seems to have no error, but there is no output of the modified firmware in the `temp` folder, and the folder is empty.And I also tried to run `build.bat` directly in cmd without modifying it , and there is also no output in the `temp` folder. 
So I'm wondering if I'm doing something wrong. I hope that any HAMs who knows how to do it can leave a message and tell me. Thank you very much.

The following is the code output in the cmd window

[image]

amnemonic, 2023-07-19:
Hello, the final modded file named `k5_v2.01.26_MODDED.bin` will be placed in the same folder as `build.bat`, not inside `temp`. Please check again :)

Noble114514, 2023-07-20:
> Hello, the final modded file named `k5_v2.01.26_MODDED.bin` will be placed in the same folder as `build.bat`, not inside `temp`. Please check again :)

Thank you for your reply. I also checked the folder where the `build.bat` is located, and there is no modified firmware in it. I even searched the entire disk with the file explorer, and there is no modified firmware found in it.
I very much hope that you can tell me where I did wrong. If you need more screenshots of the operation process or other things, please leave a message and let me know. I will continue to post them in the issue. Thank you very much for your help。

closed 2023-07-20 by Noble114514