amnemonic/Quansheng_UV-K5_Firmware #52: Multiple Language Telegram Channels

ludwich66, 2023-07-21:
I have added the following request to my mirror page. Greetings from Munich Ludwich

I would like to ask the two Spanish Telegram Channels to merge. 
It makes little sense to distribute a few OM to two "equal" channels. 
Just open a third channel and direct all users there ;-)
Here is my offer until we find a solution together!
[Telegram Channel - ES *](https://t.me/QuanShengES)
[Telegram Channel - ES #](https://t.me/Quansenguvk5)

amnemonic, 2023-07-21:
So what actually you asking me to do? 😃

ludwich66, 2023-07-22:
> So what actually you asking me to do? 😃

I have attached the mail from ea4acy which was the basis. I think you should make a peaceful contribution to avoid disputes but also create a clear structure (1 language group per country).
I will do as described. @

RAUL 2017, [21.07.2023, 16:50:16 (22.07.2023, 09:15:38)]:
Quansheng UV-K5 🇪🇸
Grupo de conversación en español del Quansheng UV-K5 y derivados con mismo formato. En este grupo no nos hacemos responsables de los firmwares modificados que los usuarios publican en este sitio.

RAUL 2017, [21.07.2023, 16:50:16 (22.07.2023, 09:15:38)]:
Hallo Jürgen.  Ich bin Raul, ea4acy.  I see in your wiki you are linking to a Telegram group for QuanSheng UV-K5 in Spanish.    Unfortunately here in Spain, after about one month  that the first telegram group was created, another user had the same idea and created a second one  😞

In both groups, the content and even the users are almost the same.  Would you be so kind to link both groups in your K5-UV wiki?   The only one you are currently linking is not the first one created, the original one, but the second one, means the one who came later to the party 😞

If you consider that only one group should be linked, i.e to match the policy with other languages groups, them let me please ask you to link the original one, not to the other who came later.     

The link for the “original one” is https://t.me/Quansenguvk5

For your info: I am no related to any of the two groups. I am just a ham user (ea4acy)  that, along with other hams, tried to explain and to convince both group’s admins about the advantages of having only one group for all spanish speaking world.  None of the Owners were interested, none of them wanted to lose “his group”. Furthermore, the owner of the second group (the one you are currently linking to) got very angry with me and banned me…  🙁

I explained this same situation to Andrej, he understood, and his GitHub is now  linking  to both groups in spanish:   The original one: https://t.me/Quansenguvk5 and the one who came later to the party: https://t.me/QuanShengES

Sorry for jump in this way to your private folder, and thanks for your time.     Ah! MANY THANKS also for your wiki, which I am recommending to any ham in Spain asking me info about QuanSheng UV-K5.

73 de Raul, ea4acy

amnemonic, 2023-07-22:
good, now there are two channels ;)

closed 2023-07-22 by amnemonic