amnemonic/Quansheng_UV-K5_Firmware #56: FM bands for hearing assistance

gabeluci, 2023-07-24:
My goal is to be able to use the UV-K5 for events that broadcast audio for hearing assistance in the range of 72-76MHz.

I have very little experience with radios. I thought I could do this out of the box, but this past weekend, I punched in the right frequency (72.9) and I got a lot of static and only barely heard what's going on amid the static. It basically wasn't usable.

I thought maybe I could use the "FM Radio" setting, but that doesn't go below 76MHz.

I think the difference is the bandwidth? I think normally it's using a bandwidth of 25kHz, but commercial FM radio uses 200kHz? But I'm pretty uneducated here, so I don't know.

So I think what I'd need is a way to extend the range of the FM Radio down to 72MHz. Is there a way to do that? I don't see that in any of the files right now.

Here's a link to the manual for a transmitter for this purpose: https://williamsav.com/wp-content/uploads/2020/04/ppa_375_manual_en.pdf

amnemonic, 2023-07-24:
Hello @gabeluci, 

inside UVK5 is chip BK1080 which can operate in these ranges:
1. 87.5–108 MHz (US/Europe, Default)
2. 76–108 MHz (Japan wide band)
3. 76–90 MHz (Japan)
4. 64-76MHz (except BK1080QB)

so I it seems that hacking firmware for radio to be able to receive band 4. will suit your needs. I can try to do this however I don't have any transmitter which works in that frequencies. Would you be able to flash and test modded firmware if prepared?

amnemonic, 2023-07-24:
Please unzip and flash below firmware if it works. It should receive FM radio in range 64-76MHz:

[k5_v2.01.26_FMRadio_64_76Mhz.bin.zip](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/files/12148501/k5_v2.01.26_FMRadio_64_76Mhz.bin.zip)

gabeluci, 2023-07-24:
Awesome. I'll flash it when I get home. I don't have a transmitter to test, but I'll see if I can find somewhere I can test it.
It's a shame it can't be changed to 64-108, but this will do since I would never actually use it for commercial FM radio anyway.

gabeluci, 2023-07-24:
I flashed it, and can confirm I can use the 64-76MHz range. I'll see if I can find an opportunity to test it out.

Thank you!

amnemonic, 2023-08-18:
Thank you too for tests!

closed 2023-08-18 by amnemonic