amnemonic/Quansheng_UV-K5_Firmware #60: downgrade firmware?

janekska, 2023-07-27:
I have UV-5R PLUS with k5_4.00.01 version firmware.
I can't flash any modded firmware - I am getting kind of version control message:

> ---------------------------
> Warning
> ---------------------------
> The file version is not match!
> ---------------------------
> OK   
> ---------------------------
> 

 I am using Portable Radio Update tools v.1.1.12

Are there any options to downgrade or how coul I get modded firmware on newer radios?

amnemonic, 2023-07-27:
Please follow this instruction [file_version_is_not_match.md](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/docs/file_version_is_not_match.md) and let me know if it helps.

janekska, 2023-07-27:
O, thanks a' lot.
I successfully flashed to k5_2.01.27.

But mostly what I am interested in are in version ...26 and then there is the same error message.

brgrds

janekska, 2023-07-27:
In previous message it is a bit wrong screenshot, the resulting mes

amnemonic, 2023-07-27:
Please attach your `k5_v2.01.26_mod_001.bin` here, I just checked exactly your scenario and it totally works.

janekska, 2023-07-27:
[k5_v2.01.26_mod_001.zip](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/files/12187317/k5_v2.01.26_mod_001.zip)

janekska, 2023-07-27:
I was able to update with this mod from telegram group:
[uv_k5_plus_26_spectrum_v58.zip](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/files/12187384/uv_k5_plus_26_spectrum_v58.zip)

amnemonic, 2023-07-27:
ok, that is really strange. `uv_k5_plus_26_spectrum_v58.zip` presents itself as `4.00.01`

what only I can suggest is following:
 - download file [qsfirm.py](https://raw.githubusercontent.com/amnemonic/Quansheng_UV-K5_Firmware/main/uvmod_kitchen/qsfirm.py)
 - place it in same folder with `k5_v2.01.26_mod_001.bin`
 - run command: `qsfirm.py unpack k5_v2.01.26_mod_001.bin fw.dec.bin fw.ver.bin`
 - edit file `fw.ver.bin` in hex editor to start with 4, for example 4.00.1
 - run command: `qsfirm.py unpack k5_v2.01.26_mod_001.bin fw.dec.bin fw.ver.bin k5_v2.01.26_mod_001_but_as_v4.bin`
 - after that you should be able to flash `k5_v2.01.26_mod_001_but_as_v4.bin` 

That also indicates that bootloader version is important for flashing procedure. At least for official update utility.

closed 2023-07-27 by janekska

TheDaylightWalker, 2023-07-28:
> Nope: [image]

I'd press Connect near COM port, before trying to flash. Updater doesn't do that automatically.

Braccato, 2023-08-01:
I have UV-5R plus I wanted to put the firmware 2.01.26 but with the procedure described here it doesn't install it

spm81, 2023-08-01:
https://github.com/spm81/Quansheng_UV-K5/tree/main/Firmware/UV-5R%20Plus

Braccato, 2023-08-01:
I want to install version 2_01_26 on UV 5R plus, any firmware does not install it

RE3CON, 2023-08-03:
> ok, that is really strange. `uv_k5_plus_26_spectrum_v58.zip` presents itself as `4.00.01`
> 
> what only I can suggest is following:
> 
> * download file [qsfirm.py](https://raw.githubusercontent.com/amnemonic/Quansheng_UV-K5_Firmware/main/uvmod_kitchen/qsfirm.py)
> * place it in same folder with `k5_v2.01.26_mod_001.bin`
> * run command: `qsfirm.py unpack k5_v2.01.26_mod_001.bin fw.dec.bin fw.ver.bin`
> * edit file `fw.ver.bin` in hex editor to start with 4, for example 4.00.1
> * run command: `qsfirm.py pack k5_v2.01.26_mod_001.bin fw.dec.bin fw.ver.bin k5_v2.01.26_mod_001_but_as_v4.bin`
> * after that you should be able to flash `k5_v2.01.26_mod_001_but_as_v4.bin`
> 
> That also indicates that bootloader version is important for flashing procedure. At least for official update utility.

this works just perfect if change the second part after editing in a Hexeditor the Version number the command from unpack to pack