amnemonic/Quansheng_UV-K5_Firmware #63: Receive on NOAA Frequ. Tabel requires a 1050 Hz Tone to demute

RE3CON, 2023-07-28:
To receive and open squelch on NOAA Frequency or replaced PMR or other freq. require a 1050 Hz Tone to open Squelch. You can use any online tone generator and transmit a 1050 Hz Tone for testing.

Is it possible to remove this requirement sending a 1050Hz tone???

I found this from NOAA receiver specs in relation:
"The 1050 Hz tone will also demute the receiver, activate a flashing LED and cause the NWS audio"
Tonegenerator: e.g.Onlinetongenerator.c o m

IK8JHL, 2023-08-17:
if it may be of interest, I created a script to modify the 1750Hz tone, if you want you can insert the 1050Hz to open the NOOA channels

RE3CON, 2023-08-18:
> if it may be of interest, I created a script to modify the 1750Hz tone, if you want you can insert the 1050Hz to open the NOOA channels

would be great. Can you share this script please?

IK8JHL, 2023-08-18:
[mod_change_Tone_1750Hz.zip](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/files/12378084/mod_change_Tone_1750Hz.zip)
 in your build.bat need to add also this line:

    python mod_change_Tone_1750Hz.py temp\fw.dec.bin

IK8JHL, 2023-08-18:
# If you want, change the frequencies of the tone burst.

tone = int(1750) # change 1750 to any frequency in Hz

##--------------------- do not modify below this line ---------------------------------------------------
import os,sys,struct
print('Running',os.path.basename(sys.argv[0]),'mod...')

fw =  bytearray(open(sys.argv[1],'rb').read())

if fw[0x29cc] == 0xd6 and fw[0x29cd] == 0x06 :
    print('Changing tone burst frequency to',tone,'Hz')
    fw[0x29cc:0x29cc+4] = struct.pack('<I',tone)
    
else:
    print('ERROR: Cant find function')

open(sys.argv[1],'wb').write(fw)

IK8JHL, 2023-08-18:
if you like follow my fork , I also added an increment for the ABR

marusc1, 2023-08-30:
Is it possible to remove the waiting for tone function, so the weather scanning could be used on changed frequencies? Without it changing the frequencies is useless.

jorgok, 2023-10-06:
Why not just disable the NOAA 1050hz receive tone?

Lar-Sen, 2023-10-16:
I disabled the need for 1050Hz toneburst while listening or scanning, for 2.01.31 version. Please check my repo.

jorgok, 2023-10-16:
I have the Quansheng UV-5R plus version. I don't see any option to disable
the 1050Hz tone burst!

Lar-Sen, 2023-10-17:
@jorgok Please have a look at mod_disable_1050hz_noaa.py in my repo. This effectively disables the need for 1050Hz toneburst. NOAA function then permits classic SQ opening, and NOAA background scanning just works like a normal scanner.

Anyway you may want to assign a different tone to each of PTT+F2 and "longpress" F1 methods : mod_change_burst_tones.py is what you need (edit toneptt and toneside variables)