amnemonic/Quansheng_UV-K5_Firmware #72: mod_piotr022_rssi_sbar unlocks TX on all bands above F3

whosmatt, 2023-08-09:
https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/7297573a468bd8da714dd43a75a626db4450e401/uvmod_kitchen/mod_piotr022_rssi_sbar.py#L17C1-L17C1

Discovered and detailed here: https://github.com/whosmatt/uvmod/issues/6

Can someone test if this happens with the uvkitchen mod as well?

lohtse, 2023-08-09:
so is it allowimg TX on more bands ?

whosmatt, 2023-08-09:
yeah, so it seems. if you could do more testing with the python mod, and
also with the compiled firmware that would be great, since this is a
somewhat dangerous bug

amnemonic, 2023-08-10:
Default firmware file with rssi bar from piotr22 have other mods included out of the box. When creating mod_piotr022_rssi_sbar I pulled only part with RSSI bar so if you don't include other mods you should only get RSSI bar.

whosmatt, 2023-08-16:
I did some testing

Stock should be
F1: 50.0000-76.0000MHz (RX only)
F2: 108-.0000-135.9750MHz (RX only)
F3: 136.0000-173.9750MHz (TX and RX both)
F4: 174.0000-349.9750MHz (RX only)
F5: 350.0000-399.9750MHz (RX only)
F6: 400.0000-469.9750MHz (TX and RX both)
F7: 470.0000-600.0000MHz (RX only)

But with RSSI firmware I got
F1: 50.0000-76.0000MHz (RX only) No TX
F2: 108-.0000-135.9750MHz (RX only) No TX
F3: 136.0000-173.9750MHz (TX and RX both) TX
**F4: 174.0000-349.9750MHz (RX only) TX
F5: 350.0000-399.9750MHz (RX only) TX**
F6: 400.0000-469.9750MHz (TX and RX both) TX
**F7: 470.0000-600.0000MHz (RX only) TX**

@piotr022

closed 2023-11-18 by amnemonic