amnemonic/Quansheng_UV-K5_Firmware #73: Errors with 2.01.27

DartRM, 2023-08-10:
With 2.01.27

RE3CON, 2023-08-18:
is there an easy way with python to make the offsets not hardcoded-constant but variable? 
search old hex string and replace with new pattern strings. Making it applicable for all fw versions.
I mean for example:
`
const offset = ?x???;
            const oldData = hexString("cf2a");
            const newData = hexString("5de0");
`

1 match

so for bands RX and TX =  5000000  -> 1800000 = ...find (oldData hexString) 40 77 1B replace with (newData hexString) 40 4B 4C
2 matches
and
2 matches 60000000 -> 130000000  string 80 A4 BF 07 etc.....
band 760000 -> 10799990 find string ..... replace string

[facsi found some offsets for v3 in his repo](https://github.com/fagci/qs-uvk5-firmware-modder/tree/main/addresses)

amnemonic, 2023-08-18:
Sure, it is possible. But is there an easy way? Probably it depends on what you will call easy 😉 You can search by offset, by bytes pattern, by function signatures, you can even [disassemble firmware on the fly](https://github.com/MythicalMeerkat/ARMv8-Disassembler-) and then patch it. Only you imagination can limit you.

RE3CON, 2023-08-18:
cool... with byte pattern search and replace it will work cause they will be always the same in every version... but not the offsets. I wish to have more time for it.

closed 2023-11-18 by amnemonic