amnemonic/Quansheng_UV-K5_Firmware #74: some python scripts test work on linux. how 2 flash use py?

BackMountainDevil, 2023-08-10:
get_version, adc_read, batt_calibrator, reboot_radio, configmem_dump, configmem_read, configmem_write are tested and work good

configmem_dump wired, may be it's that I mess the meaning of that file. I mistake it used for dump firmware.

ps: can I use a python script to flash firmware?  I tried use dnSPY to find out the address, lengeth, baudrate but failed cz I am not good at dnSPY. Good news your code shows the baudrate is 38400. [piotr022/UV_K5_playground](https://github.com/piotr022/UV_K5_playground) has a mem-map show start at 0x10C0(or Ox1000). But your docs shows flash start at 0x0F000 and end at 0x0FFFF

detail：

```bash
# K5 without battery, only with typeC
$ python version.py 
>> abcd0800026910e6b1dd58242bdfdcba
<< abcd2800036930e645a452720f05e46e2135e980f28d14e62e910d400faee76fa12d81b70b3131a05b02b861decadcba
CMD: 0x0515
LEN: 0x0024
VER: k5_2.01.26

$ python adc_read.py /dev/ttyUSB0
8.2828 V (RAW=2147)  | Charging (RAW=0)
8.2713 V (RAW=2144)  | Charging (RAW=0)
8.2751 V (RAW=2145)  | Charging (RAW=0)
8.2828 V (RAW=2147)  | Charging (RAW=0)
8.2790 V (RAW=2146)  | Charging (RAW=0)
8.2790 V (RAW=2146)  | Charging (RAW=0)
8.2713 V (RAW=2144)  | Charging (RAW=0)

$ python batt_calibrator.py /dev/ttyU
SB0 read val0
Level 0: [1299] 5.01 V // Empty battery blinking
Level 1: [1793] 6.92 V // 1 battery bar if above this value
Level 2: [1893] 7.30 V // 2 battery bars if above this value
Level 3: [1970] 7.60 V // 3 battery bars if above this value, also value used to calculate adc to volt
Level 4: [2048] 7.90 V // 4 battery bars if above this value
Level 5: [2300] 8.87 V // overwritten by radio to 2300 anyway

Actual = 8.24 V

$ python reboot_radio.py /dev/ttyUSB0

$ python configmem_dump.py /dev/ttyUS
B0 test.bin
Reading 0000...
Reading 0080...
Reading 0100...
Reading 0180...
Reading 0200...
Reading 0280...
Reading 0300...
Reading 0380...
Reading 0400...
Reading 0480...
Reading 0500...
Reading 0580...
Reading 0600...
Reading 0680...
Reading 0700...
Reading 0780...
Reading 0800...
Reading 0880...
Reading 0900...
Reading 0980...
Reading 0A00...
Reading 0A80...
Reading 0B00...
Reading 0B80...
Reading 0C00...
Reading 0C80...
Reading 0D00...
Reading 0D80...
Reading 0E00...
Reading 0E80...
Reading 0F00...
Reading 0F80...
Reading 1000...
Reading 1080...
Reading 1100...
Reading 1180...
Reading 1200...
Reading 1280...
Reading 1300...
Reading 1380...
Reading 1400...
Reading 1480...
Reading 1500...
Reading 1580...
Reading 1600...
Reading 1680...
Reading 1700...
Reading 1780...
Reading 1800...
Reading 1880...
Reading 1900...
Reading 1980...
Reading 1A00...
Reading 1A80...
Reading 1B00...
Reading 1B80...
Reading 1C00...
Reading 1C80...
Reading 1D00...
Reading 1D80...
Reading 1E00...
Reading 1E80...
Reading 1F00...
Reading 1F80...
$ md5sum test.bin 
a03555fdba7a517121f8c7132d98d6d7  test.bin
$ ls -agh test.bin 
-rw-r--r-- 1 mifen 8.0K  8月10日 11:07 test.bin

# repo download
$ md5sum ../firmware/k5_v2.01.26_publish.bin 
f7648e381d9707a43d4c2185dddf1613  ../firmware/k5_v2.01.26_publish.bin

# official download
$ md5sum k5固件v2.01.26_publish.bin 
f7648e381d9707a43d4c2185dddf1613  k5固件v2.01.26_publish.bin

$ python configmem_read.py /dev/ttyUSB0 0x0F40 8
03000000000101ff

$ python configmem_write.py /dev/ttyUSB0 0x0F40 00010001010101ff
PAYLOAD= 00010001010101ff
```

amnemonic, 2023-08-16:
> configmem_dump wired, may be it's that I mess the meaning of that file. I mistake it used for dump firmware.

yes, it is used to dump configuration memory which is in EEPROM chip, dump firmware is possible only via SWD pads on PCB.

>can I use a python script to flash firmware?

I was planning to add this option but after time I forgot about this. I've prepared [this method](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/python-utils/libuvk5.py#L160) which was intended to do the task. However finally I've decided to use official flashing app as it is safer.  

>[piotr022/UV_K5_playground](https://github.com/piotr022/UV_K5_playground) has a mem-map show start at 0x10C0(or Ox1000). But your docs shows flash start at 0x0F000 and end at 0x0FFFF

good point, when I was creating my memory map there was no possibility to "look" inside chip using SWD yet and firmware dumps, boot loader etc was obtained using modified functions running in firmware stock firmware, and the thing is that in FLASH memory of MCU boot loader is located at `0x0000` and regular firmware at `0x1000` **but** when boot process finishes then flash memory is remapped and then regular firmware is accessible at `0x0000` and bootloader at `0xF000`

closed 2023-08-17 by BackMountainDevil

bczhc, 2023-10-13:
@BackMountainDevil If you need only to find a way flashing the firmware on Linux, I just found this repo: https://github.com/sq5bpf/k5prog which does this task well for me.

I did that like this:
```shell
# fw.py is from https://github.com/fagci/qs-uvk5-firmware-modder
~/open-source/qs-uvk5-firmware-modder main ?2 ❯ ./fw.py dec ../UV-K5-Modded-Firmwares/k5_v2.01.19_mod_TX_50to850mhz.bin > ../k5prog/fw
Version: 2.01.19
~/open-source/qs-uvk5-firmware-modder main ?2 ❯ cd ../k5prog
~/open-source/k5prog main ?2 ❯ make && ./k5prog -F -YYY -b ./fw
<outputs while flashing>...
```

BackMountainDevil, 2023-10-14:
@bczhc Thank you for kindly reply. happy to know there is a way.