amnemonic/Quansheng_UV-K5_Firmware #84: Errors during font creation

mbgroot, 2023-08-24:
Hi!
I redrew the font and used the encoder. After flashing the radio, the font shows up with stripes. 
I made different pictures for the encoder - with and without color labels. The result is the same.
Large numbers were converted normally.

[image]

[mod_custom_alphabet.py.txt](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/files/12426932/mod_custom_alphabet.py.txt)

[image]

mbgroot, 2023-08-24:
Sorry, the error was in the implementation of the build.

closed 2023-08-24 by mbgroot

amnemonic, 2023-08-24:
Can you share your font though? It looks very clean and neat :)

mbgroot, 2023-08-24:
> Can you share your font though? It looks very clean and neat :)
Of course! 

Do you have an encoder for small numbers?

[image]
[image]
[mod_custom_alphabet.py.txt](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/files/12427506/mod_custom_alphabet.py.txt)
[mod_custom_bigdigits.py.txt](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/files/12427507/mod_custom_bigdigits.py.txt)
[image]

amnemonic, 2023-08-24:
> Do you have an encoder for small numbers?
Not yet but I will try to prepare one soon.

amnemonic, 2023-08-24:
@mbgroot now you can use [SmallDigits_encode.py](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/font_and_graphics/encoder/SmallDigits_encode.py)
Please let me know if it works fine :)

mbgroot, 2023-08-24:
> @mbgroot now you can use [SmallDigits_encode.py](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/font_and_graphics/encoder/SmallDigits_encode.py)
it works fine! :)
[mod_custom_smalldigits.py.txt](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/files/12427867/mod_custom_smalldigits.py.txt)