amnemonic/Quansheng_UV-K5_Firmware #85: mod_change_Tone_1750Hz.py does not actually work

whosmatt, 2023-08-26:
I could not get this mod to work, the tone when pressing the 1750Hz button is always unchanged.  
Has this ever been tested?

RE3CON, 2023-08-26:
because its wrong written but my last patch, I submitted works without a
glitch.

class Mod_ToneBurst extends FirmwareMod {
        constructor() {
            super("Repeater Tone Burst", "Changes the Tone by PTT and
Side F2 Key, used to open HAM Repeaters and NOAA Channels. The default
is 1750 Hz. To open NOAA Ton-Squelch set 1050 Hz. Other not so common
repeater tone pulse freq are 1000Hz, 1450Hz, 1750Hz, 2100Hz", 0);
            this.inputTone = addInputField(this.modSpecificDiv, "Enter
a new Tone Burst frequency (Hz)", "1050");
                    }

        apply(firmwareData) {
            const offset = 0x29cc;

            const tone = (this.inputTone.value);

            if (tone <= 0xffff) {
                const buffer = new ArrayBuffer(8);
                const dataView = new DataView(buffer);

                dataView.setUint32(0, tone, true);

                const toneHex = new Uint8Array(buffer);

                firmwareData = replaceSection(firmwareData, toneHex, offset);

                log(`Success: ${this.name} applied.`);
            }
            else {
                log(`ERROR in ${this.name}: Unexpected data, already
patched or wrong firmware?`);
            }

            return firmwareData;
        }
    }
    ,

whosmatt, 2023-08-26:
This issue is about the python mod, which does not work. Your mod will also not work, and I'm not sure why you check if tone is below 2 bytes (0xffff), just to then make an 8 byte buffer and then writing a 4 byte uint. writing these mods without a clue about data types is risky. This is the correct js implementation of the python mod: 

```js
class Mod_ChangeToneBurst extends FirmwareMod {
        constructor() {
            super("1750Hz Tone Frequency", "The 1750Hz button sends a 1750Hz activation tone by default. To open NOAA channels (in combination with the NOAA frequencies mod on the receiving unit), you can use this mod to send a 1050Hz tone. Common repeater tone pulse frequencies are 1000Hz, 1450Hz, 1750Hz, 2100Hz", 0);
            this.toneValue = addInputField(this.modSpecificDiv, "Enter a new Tone Burst value in Hz from 1000-3950:", "1750");
        }

        apply(firmwareData) {
            const minValue = 1000;
            const maxValue = 3950;
            const inputValue = parseInt(this.toneValue.value);

            if (!isNaN(inputValue) && inputValue >= minValue && inputValue <= maxValue) {
                const newData = new Uint8Array(4);
                const dataView = new DataView(newData.buffer);
                dataView.setUint32(0, inputValue, true);

                console.log(uint8ArrayToHexString(newData)); // value is correct

                firmwareData = replaceSection(firmwareData, newData, 0x29cc); // does not seem to work
                log(`Success: ${this.name} applied.`);
            }
            else {
                log(`ERROR in ${this.name}: Repeater Tone Burst must be a Tone Freq. in Hz from 1000-3950 Hz!`);
            }
            return firmwareData;
        }
    }
```
But since the python mod doesn't seem to do anything, neither will this js mod.

whosmatt, 2023-08-26:
If someone else can confirm that `mod_change_Tone_1750Hz.py` does not work it would be helpful.

wrcrooks, 2023-09-05:
Have you tried modifying the address '0x29cc' to '0x3e7c'? I find the value '39 1B' at '0x29cc' in firmware files created your UVMOD tool. Could the address have been shifted to accommodate the 'App' mods?
[image]

whosmatt, 2023-09-05:
Sadly that did not do the trick. I am taking a closer look in ghidra but I dont see an immediate problem: 

[image]
this is inside the function that processes the key presses. 

[image]
this is inside the function that resembles the "state machine" if you can really call it that. 

I am not sure why the tone gets set twice, but these are also the only values I could find and changing both of them still does not change the tone.

wrcrooks, 2023-09-05:
> https://github.com/DualTachyon/uv-k5-firmware/blob/1994d68cc5e1df36706bffca6050c6bbd1bf7a44/app/app.c#L1376
> https://github.com/DualTachyon/uv-k5-firmware/blob/1994d68cc5e1df36706bffca6050c6bbd1bf7a44/functions.c#L155

This is an amazing work effort and the community around this radio is incredible. Are you thinking of switching to the open source replication of the firmware as the base for your UVMOD site @whosmatt?

wrcrooks, 2023-09-05:
> If someone else can confirm that `mod_change_Tone_1750Hz.py` does not work it would be helpful.

I can confirm that this does not work on my stock firmware file, as the address 0x29cc contains '39 1b'. When using the mod,
`if fw[0x29cc] == 0xd6 and fw[0x29cd] == 0x06 :`
will always be false with the stock firmware and no changes will be made.

whosmatt, 2023-09-06:
> This is an amazing work effort and the community around this radio is incredible. Are you thinking of switching to the open source replication of the firmware as the base for your UVMOD site [...]

Absolutely not, because addresses and offsets change with every build. If we develop a simple filesystem with apps, I might make a tool to install apps. 
I am almost done with a installer though which pulls and installs the latest custom firmware builds.

Lar-Sen, 2023-10-16:
I adressed this issue in my own set of mods, please check my repository.
In fact I guess there is some confusion here. Initial (working) mod patched PTT+F2 key combination for 1750Hz toneburst, as many people thought the right combination was "Long press F1".
So you can have 2 possible tonebursts to configure.

closed 2023-11-18 by amnemonic