amnemonic/Quansheng_UV-K5_Firmware #87: modded firmware for UV-5R Plus

marusc1, 2023-08-28:
The modded firmware can not be loaded to Quansheng  UV-5R Plus. The error message is "The file version is not match".

RE3CON, 2023-08-28:
On UV-5R Plus is fw version 4.001 present. A version check block attemps to
"downgrade". You need to spoof after extract/decode by modded fw 2.x this
version number. replace in fw version bin file in temp folder inside the
small bin in a hex editor number 2 with 4 or just put a * instead. Continue
with build it encode together as one single bin to upload. Instructions are
here to found. Another way use uvmod select universal version build which
place a * over 2 so v4 accept it. Here you can choose custom firmware
submit/upload and only apply w/o any more patches this to flash.

marusc1, 2023-08-30:
I am trying to do what you have described and get the error:
C:\MOJE RZECZY\uvmod_kitchen_1>qsfirm.py pack temp\fw.dec.bin temp\fw.ver.bin k5_v4.01.26_publish.bin_MODDED.bin
Traceback (most recent call last):
  File "C:\MOJE RZECZY\uvmod_kitchen_1\qsfirm.py", line 70, in <module>
    decoded_firmware = open(sys.argv[2],'rb').read()
                       ^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'temp\\fw.dec.bin'
C:\MOJE RZECZY\uvmod_kitchen_1>

The files were in temp when I run buid withqsfirm.py pack.

RE3CON, 2023-08-30:
1. run
python qsfirm.py unpack k5_v4.01.26_publish.bin_MODDED.bin temp\fw.dec.bin temp\fw.ver.bin
2. edit with hexeditor in temp\fw.ver.bin
version 2. to 3 or * for universal
3. run
python qsfirm.py pack k5_v4.01.26_publish.bin_MODDED.bin temp\fw.dec.bin temp\fw.ver.bin

not try to pack a packed file. Run Unpack fist to get these two bin files.

egzumer, 2023-09-02:
create a mod_universal_version.py file with a content:

> import os,sys,struct
>print('Running',os.path.basename(sys.argv[0]),'mod...')
>
>fw =  bytearray(open(sys.argv[1],'rb').read())
>
>print('Setting universal version...')
>fw[0x0:0x0+1] = b'*'
>
>open(sys.argv[1],'wb').write(fw)

call it with:
`python mod_universal_version.py temp\fw.ver.bin`

amnemonic, 2023-09-08:
Added `mod_universal_version.py` to repo in c6b759b

closed 2023-09-08 by amnemonic