amnemonic/Quansheng_UV-K5_Firmware #88: Access to channel skip by button.

marusc1, 2023-08-28:
It would be great to have a quick access to channel skip setting when in scanning mode. Without going through a menu and changing the setting for specific channel. It is a common function in dedicated scanner radios. When scannig stops on channel stops on a channel you do not want to listen to actually because of noise on the specific frequency or uninteresting conversation you cen quickly sen this channel to be skipped in further scanning, typically but not necessary  by one touch.
Is it possible to add this function to Quansheng?

DreamATian, 2023-11-07:
push the ^button

marusc1, 2023-11-07:
> push the ^button

And it stops again on same frequency after cycling ones though the channels. The goal is to avoid it.

kittenarmy, 2023-12-03:
So a quick way to generate an on the fly ignore list, one that doesn't get remembered between power cycle.

marusc1, 2023-12-03:
So a quick way to generate an on the fly ignore list, one that doesn't get remembered between power cycles. Great description.