amnemonic/Quansheng_UV-K5_Firmware #89: TX output low on 70 cm after mods

VE6ECM, 2023-08-29:
I've noticed that my TX maximum output is only 2.4 watts on 446 Mhz after mods. Prior to mods it's 4.6watts.  I don't know how to enable the mods while retaining the higher TX output.

RE3CON, 2023-08-29:
Did you flash to K5(8)/K6 modded Firmware in version 2.01.26 over a higher
fw v2.01.27 or even v3? Or have u applied patches (they are all) for
v2.01.26 on Firmware v2.01.27 or v3.x?

VE6ECM, 2023-08-29:
Same result with all of those firmwares. However, even though the power
goes down to 2.4 watts on 446 Mhz, on 433 Mhz the power is ok at around
4.6watts. That would be ok if I was in Europe but since I'm in Canada I
operate 440 - 450. I note that with the mods, the medium power setting on
446 yields the 2.4 watts whereas the high power setting gives lower power.
Strange.

73 de VE6ECM

VE6ECM, 2023-08-29:
Funny thing is that at 435 Mhz and below the power is normal but as soon as
you go above 435 it drops.

Murray

73 de VE6ECM

radonslide, 2023-08-29:
Yes, the powerdrop happens betwenn 435 Mhz and 470Mhz.
I've just been able to replicate this with a number of mods

73 Michael

RE3CON, 2023-08-29:
please try tunas v3
https://github.com/RE3CON/UV-K5-Modded-Firmwares/blob/main/k5_V3.00.10_encrypted_18to1300MHz.bin

It should work without this issue

VE6ECM, 2023-08-29:
Ok, thanks. I did try that version without success but I'll try it again.

73 de VE6ECM

VE6ECM, 2023-08-29:
Tried again but same result.

73 de VE6ECM

RE3CON, 2023-08-29:
your radio is uv k5(8) which is the same as k6. It have orange display
background light?
55+73 RE3CON

VE6ECM, 2023-08-29:
Yes

73 de VE6ECM

VE6ECM, 2023-08-29:
After I install the FW you suggested, I get the normal power output,
however then I install the rssi mod or others and the power output problem
appears again.

73 de VE6ECM

VE6ECM, 2023-08-30:
> Thats because these patches are only for the old firmware 2.0.26. They
> work
> not like search and replace bitsets to change or add stuff on the right
> place even on different offset addresses no matter on which firmware. By
> every other firmware version these fixed programmed offset addresses will
> not more match and then it comes to overlapping or even heavier errors if
> you apply these patches for UVk5(8) onto its fw 3 or 2.00. 27. There are
> some experimental patches for firmware 2.00.27 on
> https://uvmod.valek.net.ru/
> a russian UVMod site where you can choose to use version 2.00.27, called
> fw
> 27, use a browser websie translation eg. as build in in Chrome, but
> carefully select only a few most wanted patches,. Some of them makes after
> flashing the radios LCD blank so that only the background light remains
> working and its no longer starting up but even then, the bootsector part
> remains untouched so so can easy flash it again to recover to find out
> which mod/patch function caused an error.
>
> 55s 73 RE3CON @QTH KL/MY

Ok, thanks.

73 de VE6ECM

closed 2023-10-21 by amnemonic