amnemonic/Quansheng_UV-K5_Firmware #91: Replacing or Removing Voice Audio Files [Feature Request]

wrcrooks, 2023-09-05:
I noticed that if you try to add an app to the radio in addition to a custom bootscreen, the firmware can easily end up over-sized for the memory on the radio. This got me thinking about the possibility of removing the built-in audio clips for the radio's menu "Voice" functionality to save space in the firmware. 

Would it be possible to replace these clips with an empty bit of audio then update all of the pointers to the empty file? Or identify where these clips are and replace them with custom audio files, less than or equal to them in size? I doubt this would be a simple hex edit hack and would require a complete restructure of the memory, but it could allow for multiple apps running in the firmware, right?

ANTodorov, 2023-09-26:
AFAIK the voice prompts are placed on a dedicated IC and in the firmware there are only pointers/calls when what to play... so the space savings are close to nothing.

amnemonic, 2023-11-18:
> AFAIK the voice prompts are placed on a dedicated IC and in the firmware there are only pointers/calls when what to play... so the space savings are close to nothing.

That is correct, so unless it we will not be able to reuse this memory there is no possibility to do anything about that. Closing issue for now. Thanks

closed 2023-11-18 by amnemonic