amnemonic/Quansheng_UV-K5_Firmware #94: firmware v2.01.31 - 27 Mhz frequencies shown as 118.025 Mhz

radonslide, 2023-09-16:
I tested the new mods on firmware v2.01.31.
Most of the mods are working fine for me on the UV-5R Plus.
But now all my stored CB frequencies in memory mode are listed as 118.025Mhz.

Thank you for your very good work.
I just wanted to report this as a tip

55 and 73 
Michael

egzumer, 2023-09-16:
It happens when stored channel band doesn't match the radio band range. In some moded firmwares the F2 band was 18-136MHz. It was like that because this was the only band where you could activate AM, and CB had to be in it. Now we have mod that allows AM everywhere and firmwares usually don't stuff all of those freqs to F2 (or at least yours don't). Because F2 is normally 108-136 and your stored channel is in band F2 then it is tuned to the closest possible frequency in that band.

radonslide, 2023-09-16:
Thank you for the explanation.
To test it, I entered the frequencies manually directly via the radio and then they worked again (TX/RX).
Next i try to Download these setting via CHIRP and Upload them back. See what happens.

55 and 73
Michael

egzumer, 2023-09-16:
Chirp may not work as you expect. It reads raw channel configuration and it modifies it only when you make changes to particular channel. So if you don't do any modifications it will write back anything you read exactly as is. But if you create a new CB channel entry it won't save it correctly. The driver sets all channels [<50MHz to F2 band](https://github.com/sq5bpf/uvk5-reverse-engineering/blob/main/uvk5.py#L605), and the problem repeats itself.
You can modify frequency ranges with a mod to fit those frequencies to F2 band, this will fix the problem with <50MHz channels.

radonslide, 2023-09-16:
OK. Thanks a lot for your help,
Then i used on of my radios with the .26 firmware and setting where all CB Channels are working.
The effort is then disproportionate to the result.

55 and 73
Michael

closed 2023-09-16 by radonslide