amnemonic/Quansheng_UV-K5_Firmware #104: Doppler

Xuanzai20060206, 2023-11-16:
Recently, I saw the firmware about the Doppler effect of uv-k5 on the Internet. Who has this firmware? Can you share it?
thanks！：）

var-tz, 2023-11-17:
same question! 73

amnemonic, 2023-11-18:
Hello guys! Can you give the source of this info? Also try to look around using links on the front page to other repositories, maybe there you can find these info.

var-tz, 2023-11-19:
[image]

a chinese website bilibili

var-tz, 2023-11-19:
[image]

ErofeevK, 2024-04-16:
I guess that is https://github.com/losehu/uv-k5-firmware-custom/blob/main/README_en.md

amnemonic, 2024-04-16:
Thanks @ErofeevK, I added link in  readme 8987cd9

closed 2024-04-16 by amnemonic