amnemonic/Quansheng_UV-K5_Firmware #105: Dumping k5 v2.01.32 firmware

raburton, 2023-11-18:
I have a UV K5 (8) with firmware version 2.01.32 which doesn't appear to be in your collection. I'll happily dump it, but can't find any instructions on how to do that...

amnemonic, 2023-11-18:
Hello, to dump firmware you have to be comfortable with opening the device and access these pins:
[image]

Then you have to use debugging tools compatible with OpenOCD, For example raspberry pi pico with flashed [Free-DAP](https://github.com/ataradov/free-dap) or ST-Link etc. Then use command and config form [this directory](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/tree/main/openocd) to run openocd. Then if there are no errors till this time you write in openocd console command: 
```
dump_image uv_backup.bin 0 0x1FFFF
```
If you will have any troubles during process please let us know here. This will be perfect occasion to finally write tutorial for that 😉

evyd13, 2024-06-21:
This would be needed to have an updated base firmware file for the mods, correct?