amnemonic/Quansheng_UV-K5_Firmware #109: Increase/decrease the time F key is active as modifier

kittenarmy, 2023-12-03:
I'm wondering is it possible to adjust the time F key works as modifier. The reason being I am finding the radio really slow to respond for example if I press F+3 twice in a row it doesn't recognise it as a single operation to change parameter twice. I have to press F+3, wait for the radio to change, then press it F+3 again. Maybe it is a software bug, I notice it with changing H/M/L as well.

Lar-Sen, 2023-12-03:
One thing which is slowing down responsiveness is keeping the voice prompts active. No problem when disabled.
Anyway, if you really want to experiment, it's easy to mod the F key timeout. Which firmware version are you using?

kittenarmy, 2023-12-04:
Ah thanks, I disabled it and it is quicker. Would still be nice to adjust it though. I _think_ I'm using firmware v26 from UVMOD. (Reason being I tried to patch v31 from QSFJ and it threw lots of errors which I don't recall getting before).

Lar-Sen, 2023-12-06:
As you probably know, a DTMF tone is a mix of 2 tones with different frequencies.
At firmware offset 0xA54C, you will find 8 hexadecimal 16bit values in series, which correspond to the following:

DTMF is mixed as below (tone1 is X and tone2 is Y):
```
    --------------------------------
    |	X1	X2	X3	X4 |
|----------------------------------|
|Y1 |	1	2	3	A  |
|Y2 |	4	5	6	B  |
|Y3 |	7	8	9	C  |
|Y4 |	*	0	#	D  |
------------------------------------
```

  Where:

```
-------------------------
|TONE |	HEX  |	FREQ.   |	Frequency is (HEX value)/10.32444
|-----|------|----------|
|X1   |	30C2 | 1209 Hz  |
|X2   |	35E1 | 1336 Hz  |
|X3   |	3B91 | 1477 Hz  |
|X4   |	41DC | 1633 Hz  |
|Y1   |	1C1C | 697 Hz   |
|Y2   |	1F02 | 770 Hz   |
|Y3   |	225C | 852 Hz   |
|Y4   |	25F3 | 941 Hz   |
-------------------------
```

So for example is you alter 35E1, it will change the sound of 2 , 5 , 8 and 0 at the same time.
I don't know why you want to mod DTMF tones. Is this really want you want to achieve?
By the way I don't think a python script is needed. Anyway I can craft one if you want.

Lar-Sen, 2023-12-06:
## Correct reply :)

F-key timeout is tied with other events (scanner timeout, delay to input numbers, menu timeout... and so on.
If this doesn't bother you, you can direct edit your unscrambled firmware binary like below:
`Offset 0x2638 : Value 0x10`

0x10 means 16*500ms, so 8 seconds timeout. You can decrease or increase it up to 255 (0xFF)

Let me know if you prefer a python script if ever you were not comfortable at hex editing.

Lar-Sen, 2023-12-07:
It's easily possible to make it act as the "caps lock" way: Press F to flip. So if you often use H/M/L or so, you'll have type F-6-6-F to select QRP power for example.
Another simpler fashion is to render F state sticky: It restores only if another key is pressed. No time-out.
In either case the other functions wouldn't be affected.

kittenarmy, 2023-12-13:
I found egzumer-custom does the Caps Lock, which I quite like, takes a little getting used to from stock.