amnemonic/Quansheng_UV-K5_Firmware #116: Brand new UV-K5 received this week with FW version k5_3.00.10

Mike0846178, 2023-12-25:
Hello, I'm new to two-way radios so please excuse-me if my question seems dumb.
My brand new UV-K5 has firmware k5_3.00.10 in it (yes, k5, not k6) and I can't see it in the list of this sub.
I'd like to extend the F1 range down to 18MHz but is there any mod based on the current k5_3.00.10 firmware ?
Thank you.

Lar-Sen, 2024-01-03:
As there aren't much mods for version 3.00.10, I suggest you should switch to 2.01.27 first. Your version has few differences with 2.01.27, namely horizontal menu design and ability to switch to chinese language (its main goal BTW).

Flash this https://github.com/amnemonic/Quansheng_UV-K5_Firmware/raw/main/firmware/k5_v2.01.27_flashable_on_v3.bin using Quansheng's utility, then follow usual instructions to mod it. Or just select 2.01.26/27 as a base for your firmware mod.

Lar-Sen, 2024-01-04:
Maybe because:
- many open firmware builds introduce more bugs than they resolve,
- there are users who don't mind having spectrum, MDC1200 decoding, (somewhat) better AM demodulation, and others gadgets
- some prefer cherry pick only the mods they need, customize fonts, logos and icons without requiring cross-compiler environment and know-how