amnemonic/Quansheng_UV-K5_Firmware #117: New factory FW received from QS help desk

Andy-GM, 2024-01-03:
[5RPLUS_v4.00.07_publish.zip](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/files/13818801/5RPLUS_v4.00.07_publish.zip)

Hi 

This is a copy of the latest UV-5R Plus firmware that I received from the Quansheng helpdesk. Perhaps you might wish to add it to the repository?

regards

Andy

amnemonic, 2024-01-03:
>Perhaps you might wish to add it to the repository?

Absolutely! Thanks 👍 ❤️  
FW added in commit 7d816c7

closed 2024-01-03 by amnemonic

Lar-Sen, 2024-01-04:
As I own UV-K5/K6 versions, the only feature I'd like from "new" UV-5R plus is their keypad. Orange "menu" and blue "exit" buttons are prettier.
I wish they would pack all of the best in one device..

xes, 2024-04-13:
This firmware seems to be different and the radios with bootloader 4.00.01.4 seem to be not completely compatible with other open firmwares. 

UVMOD page, using 4.00.07, produces firmwares that  doesn't boot (empty screen of the radio).

If a firmware based on 2.xx is flashed on these radios, some ranges  of frequency stop working on both reception and transmission (for example check range 150-160)

Any further feedback is welcome.