amnemonic/Quansheng_UV-K5_Firmware #119: OEM firmware change logs?

VintageEngineer, 2024-01-06:
Are there any change logs for the OEM software versions? Or even a rough descriptions?
My two UV-K5(8) units came with 2.01.32. But I see much of the mod firmware is based on 2.01.26? And now there is even 3.00.10 and 3.00.15 in the wild.
I'd love some hint at what the changes were for these OEM versions?

amnemonic, 2024-01-06:
Unfortunately there is no any change logs, absolutely nothing. Sometimes looking at reversed code it can be seen that some undocumented registers are set to different values, or some others more or less understandable changes are made. That is probably the main reason that there is no pursue to migrate lot of mods to newer versions of FW. It is worth noting that major versions like 2.xx.xx or 3.xx.xx or 4.xx.xx doesn't mean that these are better or newer. Someone at QS decided that fw 2.* is for UV-K5, 3.* is for UV-K6 and 4.* will be pushed for UV-5R PLUS. This is causing even more confusion especially that hardware-wise all these are virtually the same and it has been proven that cross-flashing doesn't do anything bad or unexpected.

VintageEngineer, 2024-01-07:
Thank you for the reply!  I'll try writing Quansheng... seems like a long long shot, but might as well try. I will report back if I get any good info back.