amnemonic/Quansheng_UV-K5_Firmware #132: How send data (audio) trought radio?

og1en, 2024-05-23:
I have K1 audio jack <-> usb A cable.
How send and receive a data from commandline linux? Or similar way from computer directly?

I need send data to other radio not setup device.

KDMcMullan, 2024-07-07:
Are you completely sure your K1 - USB cable is an "audio" cable?
It's much more likely to be a data cable.

fab1an2, 2024-07-08:
There is audio on one side ;-)

The problem is any band locks and chokes. The idea is to be able to write a simple program in C that reads binary data on the input and sends it over the radio (not over the cable but over the radio) and digital data writes out to the input. PIPE interface. 
https://reticulum.network/manual/interfaces.html

or KISS or other interface. No problem using any protocol, k17, fAX.25 or any other. Question is how to do this for reticulum. Have You any idea? Converting data from radio to computer trought sound card is pover ineficient. USB is better

KDMcMullan, 2024-07-08:
Yes, Jack plugs do not equal audio plugs. On the Kenwood K connector (arguably a pair of "audio jacks"), you have +3.3V, Gnd, Audio In, Audio Out, Serial Tx , Serial Rx, PTT.

I cannot imagine how you have a K1 USB / audio cable. What's your definition of K1? Are you talking about a K1 sound card or a Kenwood type radio connector? The cable for this radio us a USB / serial port cable. Have you made your own with a built-in USB audio soundcard?

Reference to reticulum worries me. Encrypting (what reticulum is good at) radio comms on amateur radio channels (what UV-K5 is good at) is illegal in most countries.

fab1an2, 2024-07-09:
no problem. You can use PMR band for encrypting transmission. Dont close Your mind to ham radio only.

KDMcMullan, 2024-07-09:
My mind is completely open. It is also illegal in most countries to operate the Quansheng sets on PMR frequencies.
In any case, I still don't understand the hardware (allegedly an audio cable which the poster wants to use for data).
Perhaps if the poster gave us a part number, or a web link to the hardware?

fab1an2, 2024-07-10:
reticulum.network  has some documentation. I don't. If I knew, I would have suggested diff. 
I would just want to send and receive data using quansheng hooked up to a computer. In many countries it is legal to use 433MHz and this is the majority. If someone wants to they will use on PMR.
It is important for him to have this opportunity. If there will be an interface reticulum<->quansheng<->reticulum it will be possible to see if it works. 
Connecting this is much more convenient than closed APRS. Actually reticulum has no alternative and routing data between bands.

If you can write such an interface to bring the substitute of Packet Radio back to the 21st century then help.

KDMcMullan, 2024-07-10:
If course it is perfectly legitimate to send data by ham radio with a licence. Of course people will break the law if they can. I won't be an accomplice, though.
There are perfectly simple radio data interfaces which can be purchased / built if that is the goal. A first step might be to search for "Arduino TNC." Maybe "Android TNC" as some of them can be plugged directly into a phone. I hope that helps.
Alternatively, tell me more about this audio cable and I can definitely help.

fab1an2, 2024-07-11:
**If someone wants to hit a dog, he will find a stick.**

If you think it's illegal then how about at least receiving the data and sending it to the computer? Then someone could do unlicensed listening to digital networks.

KDMcMullan, 2024-07-11:
With a very few exceptions, listening is rarely illegal.
Now, I'm prepared to help, if you tell me what your actual needs are. Otherwise ... 73.

fab1an2, 2024-07-12:
Simple read audio , convert it to digital data and send to computer using USB cable.

For example some people using digital signal to send data on band 433MHz
How receive data and print it to stdout in linux.

fAX.25, Meshtastic etc. Only convert digital data to binary format and print out.

KDMcMullan, 2024-07-12:
The HT is an analog set. In order to use it, as it stands, you would need a MODEM. In Ham land, we call it a TNC.
[og1en](https://github.com/og1en) hasn't commented in a long time. I believe the cable they have is a serial data cable. it communicates directly (not using audio) to the MCU, which communicates with the SDR. Technically, it may be possible to write a piece of firmware to convert the radio into digital. It would be a big job.

mix127, 2024-07-15:
@KDMcMullan > It would be a big job.

Radio have modem inside, sending data is probably trivial T9 example. Question how convert it to USB digital data Linux. The TNC is inside the radio. There is no need for an additional processor and additional power supply. Besides, it's a hassle and something can always break.

TNC stores data inside, there is no need for it here.
If it's a big deal and a lot of time for programmers then add it to the things to do and that's it.

KDMcMullan, 2024-07-15:
SDR has very little inside without firmware. As described, MCU is connected directly to serial port, so, firmware COULD be created to create a MODEM. If you want to see the beginnings of serial control of the radio, look to [this firmware](https://github.com/nicsure/quansheng-dock-fw). This is the closest thing I know to someone coding a serial interface to the radio. It could form the basis of a software MODEM whose serial IO is connected outside of the radio.

fab1an2, 2024-07-16:
There are people who have ideas. They throw links to articles and websites. And there are people who have ideas and implement them. I think it is not worth confusing one with the other.

I think we lack people who just know how to write something. I can't write firmweare. But I see the need to send data via radio. In an emergency situation we usually have little equipment and little time to do something.  So it is worth doing it now.

As I understand it is an idea and a need. But I don't know how to fix it/make it

KDMcMullan, 2024-07-16:
I THINK I found what we need.