amnemonic/Quansheng_UV-K5_Firmware #139: Inquiry About Differences and Compatibility Between Bootloader Versions

z1n8tjshh, 2024-10-31:
I noticed that there are several versions of the bootloader files available in the repository:

    boot_v2.00.06.bin
    boot_v3.00.02.bin
    boot_v4.00.01.bin

After using the following command to dump my own bootloader:

`openocd -f interface/stlink.cfg -f target/dp32g030.cfg -c "reset halt" -c "dump_image bootloader.bin 0 0x1000" -c "shutdown"`

I found that the content matches boot_v2.00.06.bin, my device is QuanSheng K6(Mainland China version)

I would like to know the following:

1. What are the differences between these versions? Do they have functional differences?
2. Are they intended for different devices or regions, or do they represent older/newer versions?
3. Are these bootloader versions interchangeable, or are there compatibility issues?
4. What is the source of these files? Were they dumped from different devices or provided by QuanSheng officially?

Any insights or documentation regarding these differences and compatibility would be greatly appreciated.

Thank you!

amnemonic, 2024-10-31:
Hello!

I'll try to answer your questions, but please note that some information might be outdated since I'm no longer actively involved in development.

**Regarding your questions:**

 `1`. & `2`. The bootloader code is primarily used for flashing and factory calibration. The main difference lies in the firmware compatibility:

- `boot_v2` accepts `k5_v2*` (UV-K5) firmware.
- `boot_v3` accepts only `k5_v3*` (UV-K6) firmware, and so on. 

 `3`. All versions should be interchangeable, but I don't expect any noticeable changes resulting from switching between them.

 `4`.  All these versions were extracted using `openocd`, just like you did, from different radios.

While QuanSheng could potentially release firmware with updated bootloaders, to my knowledge this hasn't happened (yet). There are no official (and likely no unofficial) documentation resources for bootloader firmware. 

Did your K6 came with Chinese menu? If not then this would explain why it has `boot_v2` and not `v3` as we would expect.

z1n8tjshh, 2024-11-01:
Thank you for your reply! My device does indeed have a Chinese menu, but its boot version is still v2.

When I tried to flash the official v3.00.18, I encountered an error indicating an incorrect version. I resolved this issue using qsfirm and mod_universal_version in uvmod_kitchen. If anyone encounters a similar problem, can also use the following commands:

```
# Unpack
python qsfirm.py unpack K6_ms_lpf32_v3.00.18_publish.bin fw.dec.bin fw.ver.bin
# Modify
python mod_universal_version.py fw.ver.bin
# Repack
python qsfirm.py pack fw.dec.bin fw.ver.bin Fixed_K6_ms_lpf32_v3.00.18_publish.bin
```

I'm wondering if:

1. Is it possible that my device is equipped with a 64KB (0.5Mbit) EEPROM, while the newer 512KB (4Mbit) version is the one that comes with boot_v3?
2. Could it be that a fraudulent seller modified a K5 to look like a K6 by changing the casing and then flashed the K6 Chinese menu firmware onto it, without updating the boot version? Given that there are no significant hardware differences between the K5 and K6, this would be relatively easy to do.

Thank you for your help!

z1n8tjshh, 2024-11-01:
> theres a new version???"while the newer 512KB (4Mbit) version" or is this just the DIY board???

There is a limegreeen version, came with 512KB (4Mbit). 
see: 
https://www.reddit.com/r/Quansheng/comments/1d8hyi7/so_the_limegreen_uvk6_has_a_larger_memory_than/
https://post.smzdm.com/p/apm2z3m7/

but yeah, alot of us just replace the factory eeprom with a DIY board+2*M24M02, or just stack 2 on each other, cost about 5 CNY(0.70 USD) for the parts. 
see: https://www.bilibili.com/read/cv36560294

z1n8tjshh, 2024-11-01:
> > > theres a new version???"while the newer 512KB (4Mbit) version" or is this just the DIY board???
> > 
> > 
> > There is a limegreeen version, came with 512KB (4Mbit). see: https://www.reddit.com/r/Quansheng/comments/1d8hyi7/so_the_limegreen_uvk6_has_a_larger_memory_than/ https://post.smzdm.com/p/apm2z3m7/
> > but yeah, alot of us just replace the factory eeprom with a DIY board+2*M24M02, or just stack 2 on each other, cost about 5 CNY(0.70 USD) for the parts. see: https://www.bilibili.com/read/cv36560294
> 
> Arrr ok just what I thought
> 
> Will wait till the offical one is released... and apologies for possibly hijacking your thread..

the limegreen one is officially released by QuanSheng, but only in China i think, it's in their official online store in TaoBao & JD(Consider as China's Amazon).