amnemonic/Quansheng_UV-K5_Firmware #140: Dual Watch (TDR), Rx Priority, Secondary Mute

ryansavino, 2025-03-09:
I had a question here about dual watch functionality. I'm trying to setup 2 channels for rx dual watch:

- 1st channel is for primary communication rx and tx
- 2nd channel is for just receive, but will have continuous communication (like a NASCAR race broadcast)

This setup is not working for me, and after researching a bit, I think I know what's happening.

First off, I think the default TDR functionality is to switch tx onto whichever channel is receiving. The second channel above was being selected because of the continual broadcast. There's a way to override this in the TDR menu setting by setting the tx primary to a specific channel A or B.

I think the real functionality I'm looking for is rx priority. This would set rx on channel A to always override channel B. After googling a bit online, I've seen some different radios with a feature for secondary or sub channel mute. I believe this functionality would actually mute or dampen the rx on the second broadcast channel when there was rx on the primary first channel.

Anyone here familiar with these features or where to go about making modifications to add or support these features? I'd be willing to dive in myself, but it would help if someone could point me in the right direction. Thanks in advance.

madflytom, 2025-03-09:
Hey Ryan!

Have you looked at menu 55?  I think the setting you are looking for is:

MAIN TX DUAL RX - always transmits on the primary, listens to both

This would allow you to put your transmit frequency on the main VFO, and the other frequency will just listen.  

I haven’t tested it but I think “main” is whatever frequency is selected when you press the a/b button.  So set your transmit frequency as main and the dual watch will switch to the secondary frequency when it gets a signal.  If you press PTT it will go out on the main frequency.

Something else you can try is setting the scan resume mode timer to longer, so after you transmit it takes longer to switch back to the RX only channel.  With dual watch, I think as long as you have a signal on the RX frequency the primary isn’t being monitored.

madflytom, 2025-03-09:
Another option might be set both channels in a scan bank, and set the scnrev setting to timeout.  Then also set the transmit channel to VFO B, and set one of the multi-function buttons to be a PTT button.  

This would allow you to scan both channels, and when one has traffic it would stay there until the timer expires and starts the scan again.  You’d hear any traffic on the other channel during that scan, and it would stay there until timeout.  The PTT on VFO b would let you transmit on that channel by effectively cancelling the scan.  

Not sure if any of that would work in practice but maybe worth a try!